Seeed-Studio/Seeed_Arduino_LSM6DS3 #4: Versioning issue

FarFlyer, 2021-12-03:
### Versioning issue
There is a little bit of misalignment in the versioning.

Please make a release with tag 2.0.1 or higher.

Here on GitHub the latest release is 1.0.1.

On the [Arduino website](https://www.arduino.cc/reference/en/libraries/accelerometer-and-gyroscope-lsm6ds3/) there is the version 2.0.0.

In platformIO the latest is 2.0.0, but the release note shows 1.0.1

Those three versions are different from one to another.

Please make the proper changes and cascade the info.
[image]
[image]
[image]

Pillar1989, 2022-01-20:
@lakshanthad  @0hotpotman0  Please support this issue, and make this repo go to the Arduino library manager.

lakshanthad, 2022-01-20:
Hello,

I have released v2.0.1 and also asked Arduino to merge the changes to the repo name accordingly.
https://github.com/arduino/library-registry/issues/929

Please wait a while for the changes to reflect @FarFlyer

closed 2022-02-09 by Pillar1989