Seeed-Studio/Seeed_Arduino_LSM6DS3 #7: accelerometer not at ZERO at rest

debungle-it, 2022-06-02:
I am running the HighLevelExample on a XIAO Sense. Even at rest, I get significant outputs:

11:53:34.842 -> Device OK!
11:53:34.842 ->  X1 = 0.0000 Y1 = 0.2743 Z1 = 0.1864
11:53:35.856 ->  X1 = -0.7149 Y1 = 0.5929 Z1 = 0.4055
11:53:36.831 ->  X1 = -0.7164 Y1 = 0.5954 Z1 = 0.4065
11:53:37.844 ->  X1 = -0.7149 Y1 = 0.5963 Z1 = 0.4070
11:53:38.851 ->  X1 = -0.7159 Y1 = 0.5958 Z1 = 0.4060
11:53:39.832 ->  X1 = -0.7193 Y1 = 0.5963 Z1 = 0.4085
11:53:40.845 ->  X1 = -0.7193 Y1 = 0.5958 Z1 = 0.4065
11:53:41.859 ->  X1 = -0.7164 Y1 = 0.5963 Z1 = 0.4060
11:53:42.839 ->  X1 = -0.7144 Y1 = 0.5973 Z1 = 0.4055
11:53:43.855 ->  X1 = -0.7159 Y1 = 0.5968 Z1 = 0.4070
11:53:44.855 ->  X1 = -0.7178 Y1 = 0.5978 Z1 = 0.4046
11:53:45.852 ->  X1 = -0.7154 Y1 = 0.5963 Z1 = 0.4055
11:53:46.877 ->  X1 = -0.7178 Y1 = 0.5949 Z1 = 0.4055
11:53:47.864 ->  X1 = -0.7159 Y1 = 0.6554 Z1 = 0.4085

Depending on the run, the values can be over 1G. Is there a way to calibrate the IMU, or do I need to measure the initial offset and subtract those values from the "real" readings?

lakshanthad, 2022-06-09:
Hello,

I have tested this on a XIAO BLE Sense with the onboard LSM6DS IMU and the results are as follows:

Example Code:
https://github.com/Seeed-Studio/Seeed_Arduino_LSM6DS3/blob/master/examples/HighLevelExample/HighLevelExample.ino

Results:

- When the accelerometer is stationary on X-axis:

```sh
X1 = 1.0146 Y1 = -0.0776 Z1 = 0.0078
X1 = 1.0160 Y1 = -0.0776 Z1 = 0.0088
X1 = 1.0141 Y1 = -0.0771 Z1 = 0.0078
X1 = 1.0136 Y1 = -0.0781 Z1 = 0.0068
X1 = 1.0155 Y1 = -0.0766 Z1 = 0.0093
X1 = 1.0155 Y1 = -0.0791 Z1 = 0.0073
X1 = 1.0136 Y1 = -0.0781 Z1 = 0.0068
X1 = 1.0146 Y1 = -0.0781 Z1 = 0.0073
X1 = 1.0146 Y1 = -0.0766 Z1 = 0.0073
X1 = 1.0150 Y1 = -0.0771 Z1 = 0.0073
X1 = 1.0141 Y1 = -0.0771 Z1 = 0.0054
X1 = 1.0155 Y1 = -0.0771 Z1 = 0.0068
X1 = 1.0165 Y1 = -0.0776 Z1 = 0.0049
X1 = 1.0136 Y1 = -0.0766 Z1 = 0.0059
X1 = 1.0121 Y1 = -0.0786 Z1 = 0.0063
```

- When the accelerometer is stationary on Y-axis:

```sh
X1 = -0.0161 Y1 = 0.9926 Z1 = 0.0527
X1 = -0.0117 Y1 = 0.9941 Z1 = 0.0434
X1 = -0.0102 Y1 = 0.9941 Z1 = 0.0522
X1 = -0.0107 Y1 = 0.9911 Z1 = 0.0625
X1 = -0.0098 Y1 = 0.9916 Z1 = 0.0464
X1 = -0.0117 Y1 = 0.9941 Z1 = 0.0425
X1 = -0.0098 Y1 = 0.9955 Z1 = 0.0444
X1 = -0.0132 Y1 = 0.9936 Z1 = 0.0444
X1 = -0.0107 Y1 = 0.9926 Z1 = 0.0429
X1 = -0.0107 Y1 = 0.9906 Z1 = 0.0434
X1 = -0.0127 Y1 = 0.9926 Z1 = 0.0420
X1 = -0.0112 Y1 = 0.9945 Z1 = 0.0473
X1 = -0.0112 Y1 = 0.9936 Z1 = 0.0464
X1 = -0.0107 Y1 = 0.9926 Z1 = 0.0449
X1 = -0.0102 Y1 = 0.9960 Z1 = 0.0371
```

- When the accelerometer is stationary on Z-axis:

```sh
X1 = -0.0464 Y1 = -0.0244 Z1 = 0.9950
X1 = -0.0444 Y1 = -0.0244 Z1 = 0.9970
X1 = -0.0429 Y1 = -0.0259 Z1 = 0.9960
X1 = -0.0420 Y1 = -0.0293 Z1 = 0.9975
X1 = -0.0395 Y1 = -0.0298 Z1 = 0.9955
X1 = -0.0425 Y1 = -0.0254 Z1 = 0.9960
X1 = -0.0449 Y1 = -0.0273 Z1 = 0.9970
X1 = -0.0439 Y1 = -0.0244 Z1 = 0.9980
X1 = -0.0429 Y1 = -0.0234 Z1 = 0.9970
X1 = -0.0415 Y1 = -0.0254 Z1 = 0.9965
X1 = -0.0439 Y1 = -0.0249 Z1 = 0.9970
X1 = -0.0425 Y1 = -0.0264 Z1 = 0.9984
X1 = -0.0439 Y1 = -0.0264 Z1 = 0.9970
X1 = -0.0415 Y1 = -0.0259 Z1 = 0.9965
X1 = -0.0439 Y1 = -0.0263 Z1 = 0.9971
```

It seems that in each orientation, the corresponding values for the axis responsible for that orientation are closing to 1G which is correct.

Do you get these kinds of results as well?

torkfat, 2022-06-09:
Hi
I appreciate your help. I have another question. Does the data that comes out of readFloatAccelX have any axis fusion done to it?

Lesords, 2024-10-16:
Hello,

I'm very sorry to have kept you waiting so long.

May I ask if you have solved this problem so far?

closed 2024-10-30 by Lesords