Seeed-Studio/Seeed_Arduino_LSM6DS3 #13: Warnings

marcusobrien, 2023-07-31:
Using latest Arduino IDE 2.1.0
And toolchain for Seeed XIAO nRF52840

Many warnings in code - if warnings are set as errors, these fail the compilation

d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp: In member function 'status_t LSM6DS3Core::readRegisterRegion(uint8_t*, uint8_t, uint8_t)':
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:160:13: warning: unused variable 'tempFFCounter' [-Wunused-variable]
  160 |     uint8_t tempFFCounter = 0;
      |             ^~~~~~~~~~~~~
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp: In member function 'status_t LSM6DS3Core::readRegister(uint8_t*, uint8_t)':
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:264:20: warning: 'result' may be used uninitialized in this function [-Wmaybe-uninitialized]
  264 |     *outputPointer = result;
      |     ~~~~~~~~~~~~~~~^~~~~~~~
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp: In member function 'void LSM6DS3::fifoClear()':
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:827:25: warning: 'result' may be used uninitialized in this function [-Wmaybe-uninitialized]
  827 |     tempAccumulator |= (tempReadByte << 8);
      |                         ^~~~~~~~~~~~
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:224:13: note: 'result' was declared here
  224 |     uint8_t result;
      |             ^~~~~~
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:827:21: warning: 'result' may be used uninitialized in this function [-Wmaybe-uninitialized]
  827 |     tempAccumulator |= (tempReadByte << 8);
      |     ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:224:13: note: 'result' was declared here
  224 |     uint8_t result;
      |             ^~~~~~
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp: In member function 'float LSM6DS3::readFloatAccelX()':
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:280:41: warning: 'myBuffer[0]' may be used uninitialized in this function [-Wmaybe-uninitialized]
  280 |     int16_t output = (int16_t)myBuffer[0] | int16_t(myBuffer[1] << 8);
      |                               ~~~~~~~~~~^
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:280:63: warning: 'myBuffer[1]' may be used uninitialized in this function [-Wmaybe-uninitialized]
  280 |     int16_t output = (int16_t)myBuffer[0] | int16_t(myBuffer[1] << 8);
      |                                                     ~~~~~~~~~~^
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:280:41: warning: 'myBuffer[0]' may be used uninitialized in this function [-Wmaybe-uninitialized]
  280 |     int16_t output = (int16_t)myBuffer[0] | int16_t(myBuffer[1] << 8);
      |                               ~~~~~~~~~~^
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:280:63: warning: 'myBuffer[1]' may be used uninitialized in this function [-Wmaybe-uninitialized]
  280 |     int16_t output = (int16_t)myBuffer[0] | int16_t(myBuffer[1] << 8);
      |                                                     ~~~~~~~~~~^
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp: In member function 'float LSM6DS3::readFloatAccelZ()':
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:280:41: warning: 'myBuffer[0]' may be used uninitialized in this function [-Wmaybe-uninitialized]
  280 |     int16_t output = (int16_t)myBuffer[0] | int16_t(myBuffer[1] << 8);
      |                               ~~~~~~~~~~^
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:280:63: warning: 'myBuffer[1]' may be used uninitialized in this function [-Wmaybe-uninitialized]
  280 |     int16_t output = (int16_t)myBuffer[0] | int16_t(myBuffer[1] << 8);
      |                                                     ~~~~~~~~~~^
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:280:41: warning: 'myBuffer[0]' may be used uninitialized in this function [-Wmaybe-uninitialized]
  280 |     int16_t output = (int16_t)myBuffer[0] | int16_t(myBuffer[1] << 8);
      |                               ~~~~~~~~~~^
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:280:63: warning: 'myBuffer[1]' may be used uninitialized in this function [-Wmaybe-uninitialized]
  280 |     int16_t output = (int16_t)myBuffer[0] | int16_t(myBuffer[1] << 8);
      |                                                     ~~~~~~~~~~^
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp: In member function 'float LSM6DS3::readFloatAccelY()':
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:280:41: warning: 'myBuffer[0]' may be used uninitialized in this function [-Wmaybe-uninitialized]
  280 |     int16_t output = (int16_t)myBuffer[0] | int16_t(myBuffer[1] << 8);
      |                               ~~~~~~~~~~^
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:280:63: warning: 'myBuffer[1]' may be used uninitialized in this function [-Wmaybe-uninitialized]
  280 |     int16_t output = (int16_t)myBuffer[0] | int16_t(myBuffer[1] << 8);
      |                                                     ~~~~~~~~~~^
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:280:41: warning: 'myBuffer[0]' may be used uninitialized in this function [-Wmaybe-uninitialized]
  280 |     int16_t output = (int16_t)myBuffer[0] | int16_t(myBuffer[1] << 8);
      |                               ~~~~~~~~~~^
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:280:63: warning: 'myBuffer[1]' may be used uninitialized in this function [-Wmaybe-uninitialized]
  280 |     int16_t output = (int16_t)myBuffer[0] | int16_t(myBuffer[1] << 8);
      |                                                     ~~~~~~~~~~^
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp: In member function 'float LSM6DS3::readFloatGyroZ()':
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:280:41: warning: 'myBuffer[0]' may be used uninitialized in this function [-Wmaybe-uninitialized]
  280 |     int16_t output = (int16_t)myBuffer[0] | int16_t(myBuffer[1] << 8);
      |                               ~~~~~~~~~~^
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:280:63: warning: 'myBuffer[1]' may be used uninitialized in this function [-Wmaybe-uninitialized]
[... 118 lines cut ...]
      |                         ^~~~~~~~~~~~~~~~~~~~~~
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp: In member function 'uint16_t LSM6DS3::fifoGetStatus()':
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:827:21: warning: 'result' may be used uninitialized in this function [-Wmaybe-uninitialized]
  827 |     tempAccumulator |= (tempReadByte << 8);
      |     ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:827:25: warning: 'result' may be used uninitialized in this function [-Wmaybe-uninitialized]
  827 |     tempAccumulator |= (tempReadByte << 8);
      |                         ^~~~~~~~~~~~
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp: In member function 'status_t LSM6DS3Core::beginCore()':
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:130:9: warning: 'result' may be used uninitialized in this function [-Wmaybe-uninitialized]
  130 |     if (!(readCheck == LSM6DS3_ACC_GYRO_WHO_AM_I || readCheck == LSM6DS3_C_ACC_GYRO_WHO_AM_I)) {
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp: In member function 'status_t LSM6DS3::begin()':
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:482:17: warning: 'result' may be used uninitialized in this function [-Wmaybe-uninitialized]
  482 |     dataToWrite &= ~((uint8_t)LSM6DS3_ACC_GYRO_BW_SCAL_ODR_ENABLED);
      |     ~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:130:9: warning: 'result' may be used uninitialized in this function [-Wmaybe-uninitialized]
  130 |     if (!(readCheck == LSM6DS3_ACC_GYRO_WHO_AM_I || readCheck == LSM6DS3_C_ACC_GYRO_WHO_AM_I)) {
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:224:13: note: 'result' was declared here
  224 |     uint8_t result;
      |             ^~~~~~
d:\Arduino\libraries\Seeed_Arduino_LSM6DS3\LSM6DS3.cpp:553:16: warning: 'result' may be used uninitialized in this function [-Wmaybe-uninitialized]
  553 |         } else if (result == LSM6DS3_C_ACC_GYRO_WHO_AM_I) { //0x6A LSM6dS3-C
      |                ^~

Lesords, 2024-10-17:
Hello,

I'm very sorry to have kept you waiting so long.

Do you still have this problem now?

is-qian, 2024-12-18:
Thank you for using this repository. In this [commit](https://github.com/Seeed-Studio/Seeed_Arduino_LSM6DS3/commit/1c14d8c27a521136e74509853a408b167e6dee65), these warnings have been resolved. You can try again with the latest version of the repository. Thank you once again for your contribution.

closed 2024-12-26 by is-qian