Seeed-Studio/Seeed_Arduino_LSM6DS3 #14: FreeFallDetect sample doesn't work

TobiasReich, 2023-08-29:
I tried the sample for the free fall detection but it doesn't seem to work. No fall is detected.

Maybe this sample code could be updated. Or am I missing something?

TobiasReich, 2023-09-04:
Turns out there is a bug in the sample code for the free fall detection.

error += lsm6ds3.writeRegister(LSM6DS3_ACC_GYRO_TAP_CFG1, 0x01);

has to be actually

error += lsm6ds3.writeRegister(LSM6DS3_ACC_GYRO_TAP_CFG1, 0x81);

A better description can be found here:

https://forum.seeedstudio.com/t/xiao-nrf52840-free-fall-detection-not-working/272103/9

TobiasReich, 2023-09-04:
I added a pull request.
https://github.com/Seeed-Studio/Seeed_Arduino_LSM6DS3/pull/15
Maybe someone might be so kind to accept it!
Thanks!

Lesords, 2024-10-16:
Hello,

I'm very sorry to have kept you waiting so long.

The PR you mention have been merged so far, I'm about to close this issue, you're welcome to reopen the issue if you have any other questions.

closed 2024-10-16 by Lesords