Seeed-Studio/Seeed_Arduino_LSM6DS3 #16: Please provide more information about example IMU_Classifier.ino

brightproject, 2023-11-27:
Hello @lakshanthad.
Could you provide more information about the `tensoflow` example?
What work does the code do, what is the `model.h` file and how to create it?

Lesords, 2024-10-16:
Hello,

I'm very sorry to have kept you waiting so long.

May I ask if you have solved this problem so far?

brightproject, 2024-10-16:
Hello @Lesords 
Yes, the question is still relevant🙂

Lesords, 2024-10-16:
I don't have the hardware for it at the moment, but when I do I'll check it out and let you know.

is-qian, 2024-12-18:
Sorry to keep you waiting. Regarding the use of TensorFlow and model generation, you can find the entire process here. We welcome your use and testing of it.

is-qian, 2024-12-18:
Sorry to keep you waiting. Regarding the use of TensorFlow and model generation, you can find the entire process [here](https://wiki.seeedstudio.com/XIAO-BLE-Sense-TFLite-Getting-Started/#inference). We welcome your use and testing of it.

closed 2024-12-26 by is-qian