Seeed-Studio/Seeed_Arduino_LSM6DS3 #20: Bug - 'SPI_CLOCK_DIV4' was not declared in this scope against Xiao MG24 Sense

rei-vilo, 2024-12-17:
Compiling the example provided at https://wiki.seeedstudio.com/xiao_mg24_sense_built_in_sensor/#code-implementation against Xiao MG24 Sense raises the error 

> 'SPI_CLOCK_DIV4' was not declared in this scope.

```
/home/USER/.arduino15/packages/SiliconLabs/tools/gcc-arm-none-eabi/12.2.rel1/bin/arm-none-eabi-g++ -mcpu=cortex-m33 -mthumb -mfpu=fpv5-sp-d16 -mfloat-abi=hard -c -g -w -std=c++11 -std=gnu++17 -Wall -Wextra -fno-exceptions -fdata-sections -ffunction-sections -fomit-frame-pointer -imacros sl_gcc_preinclude.h -mcmse -Wno-deprecated-declarations -Wno-maybe-uninitialized -Wno-missing-field-initializers -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -fno-strict-aliasing -fno-unwind-tables -fno-asynchronous-unwind-tables -fno-common -Wno-sign-compare --specs=nano.specs -g -Os -DF_CPU=39000000 -DARDUINO=10607 "-DARDUINO_SILABS=\"2.2.0\"" -DARDUINO_XIAO_MG24 -DARDUINO_ARCH_SILABS -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/cores/silabs/api/deprecated -DNUM_LEDS=1 -DNUM_HW_SERIAL=2 -DNUM_HW_SPI=2 -DNUM_HW_I2C=2 -DNUM_DAC_HW=2 -DARDUINO_MAIN_TASK_STACK_SIZE=2048 -DARDUINO_MATTER -DCHIP_CRYPTO_PLATFORM=1 -DIS_DEMO_LIGHT=1 -DNVM3_DEFAULT_MAX_OBJECT_SIZE=4092 -DNVM3_DEFAULT_NVM_SIZE=40960 -DSL_STATUS_LED=0 -D_WANT_REENT_SMALL=1 -DconfigNUM_THREAD_LOCAL_STORAGE_POINTERS=2 -DconfigNUM_USER_THREAD_LOCAL_STORAGE_POINTERS=0 -DCHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=<lib/address_resolve/AddressResolve_DefaultImpl.h> -DCHIP_HAVE_CONFIG_H=1 -DRADIO_CONFIG_DMP_SUPPORT=1 -DCURRENT_TIME_NOT_IMPLEMENTED=1 -DMBEDTLS_USER_CONFIG_FILE=<sli_psa_builtin_config.h> -DOPENTHREAD_CONFIG_DETERMINISTIC_ECDSA_ENABLE=0 -DOPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS=0 -DSILABS_OTA_ENABLED=1 -DRTT_USE_ASM=0 -DENABLE_WSTK_LEDS=1 -DEFR32MG24B220F1536IM48=1 -DSL_APP_PROPERTIES=1 -DHARDWARE_BOARD_DEFAULT_RF_BAND_2400=1 -DHARDWARE_BOARD_SUPPORTS_1_RF_BAND=1 -DHARDWARE_BOARD_SUPPORTS_RF_BAND_2400=1 -DHFXO_FREQ=39000000 "-DSL_BOARD_NAME=\"BRD4187C\"" "-DSL_BOARD_REV=\"A01\"" -DconfigNUM_SDK_THREAD_LOCAL_STORAGE_POINTERS=2 -DSL_COMPONENT_CATALOG_PRESENT=1 -DMBEDTLS_CONFIG_FILE=<sl_mbedtls_config.h> "-DOPENTHREAD_CORE_CONFIG_PLATFORM_CHECK_FILE=\"openthread-core-efr32-config-check.h\"" "-DOPENTHREAD_PROJECT_CORE_CONFIG_FILE=\"openthread-core-efr32-config.h\"" "-DOPENTHREAD_CONFIG_FILE=\"sl_openthread_generic_config.h\"" -DOPENTHREAD_FTD=1 "-DSL_OPENTHREAD_STACK_FEATURES_CONFIG_FILE=\"sl_openthread_features_config.h\"" -DMBEDTLS_PSA_CRYPTO_CONFIG_FILE=<psa_crypto_config.h> -DSL_RAIL_LIB_MULTIPROTOCOL_SUPPORT=1 -DSL_RAIL_UTIL_PA_CONFIG_HEADER=<sl_rail_util_pa_config.h> -DSLI_RADIOAES_REQUIRES_MASKING=1 -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/autogen/zap-generated -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/autogen/zap-generated/app -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/config -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/config/btconf -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/config/common -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/autogen -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/include -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/examples/lighting-app/silabs/include -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/examples/platform/silabs -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/include -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/lib -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/third_party/nlassert/repo/include -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/third_party/nlio/repo/include -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/zzz_generated/app-common -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/lib/address_resolve -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/app -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/zzz_generated/app-common/app-common/zap-generated/attributes -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/lib/dnssd -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/examples/platform/silabs/efr32 -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/platform/silabs/efr32 -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/include/platform -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/include/platform/internal -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/platform/logging -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/include/platform/internal/testing -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/platform/silabs -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/platform/OpenThread -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/platform/silabs/platformAbstraction -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/platform -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/platform/FreeRTOS -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/protocols/interaction_model -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/access/examples -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/access -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/lib/asn1 -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/protocols/bdx -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/ble -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/app/server -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/controller -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/lib/core -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/crypto -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/app/reporting -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/app/MessageDef -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/protocols/echo -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/protocols/user_directed_commissioning -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/protocols -I/home/USER/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/variants/xiao_mg24/matter/matter_2.2.0/src/protocols/secure_channel -I/home/USER/.ardui [... cut ...]
/home/USER/Projets/Arduino/libraries/Seeed_Arduino_LSM6DS3/LSM6DS3.cpp: In member function 'status_t LSM6DS3Core::beginCore()':
/home/USER/Projets/Arduino/libraries/Seeed_Arduino_LSM6DS3/LSM6DS3.cpp:89:33: error: 'SPI_CLOCK_DIV4' was not declared in this scope
   89 |             SPI.setClockDivider(SPI_CLOCK_DIV4);
      |                                 ^~~~~~~~~~~~~~

Using library Seeed Arduino LSM6DS3 at version 2.0.3 in folder: /home/USER/Projets/Arduino/libraries/Seeed_Arduino_LSM6DS3 

```

punkweazel, 2024-12-17:
Same problem here...
I'm working on a sketch that compiles fine for the Seeed XIAO nRF52840 Sense.  Changing the board to MG24 Sense, and I get this error.

rei-vilo, 2024-12-17:
One more problem for the example provided at https://wiki.seeedstudio.com/xiao_mg24_sense_built_in_sensor/#code-implementation against Xiao MG24 Sense

> Device error

because according to the [schematics](https://files.seeedstudio.com/wiki/XIAO_MG24/Getting_Start/XIAO_MGM240S_KICAD_Prj.pdf), the IMU is connected to pins PB2 PB3 D14 and not default Wire on pins PC4 PC5.

is-qian, 2024-12-18:
Thank you for your verification. In this [commit](https://github.com/Seeed-Studio/Seeed_Arduino_LSM6DS3/commit/a4a046cec9b1c5dfea994884e521194dabdb29d0), the error has been addressed. Please try again with the latest library. Feel free to raise any issues promptly. Thank you once again.

rei-vilo, 2024-12-18:
Solved with the commit, thank you!

closed 2024-12-18 by rei-vilo