Seeed-Studio/Seeed_Arduino_LSM6DS3 #21: Bug - Library does not support Xiao MG24 Sense

rei-vilo, 2024-12-17:
One more problem for the example provided at https://wiki.seeedstudio.com/xiao_mg24_sense_built_in_sensor/#code-implementation against Xiao MG24 Sense.

> Device error

because according to the [schematics](https://files.seeedstudio.com/wiki/XIAO_MG24/Getting_Start/XIAO_MGM240S_KICAD_Prj.pdf), the IMU is connected to pins PB2 PB3 and not default Wire on pins PC4 PC5.

A special condition should be added for the Xiao MG24.

https://github.com/Seeed-Studio/Seeed_Arduino_LSM6DS3/blob/a4a046cec9b1c5dfea994884e521194dabdb29d0/LSM6DS3.cpp#L35-L37

Trying `TARGET_SEEED_XIAO_MG24_SENSE` does not work.

is-qian, 2024-12-18:
As you mentioned, ARDUINO_SILABS is indeed the macro. If you are unable to read the IMU readings, it might be because you have not enabled the PD5 pin, which is the IMU enable pin added for power-saving considerations. For specific code examples, you can refer to the [wiki](https://wiki.seeedstudio.com/xiao_mg24_sense_built_in_sensor/#code-implementation).

is-qian, 2024-12-18:
Just as you mentioned, ARDUINO_SILABS is the macro you were referring to. If you're not getting any IMU readings, I suspect it might be because you haven't enabled the PD5 pin, which is an IMU enable pin added for the purpose of low power consumption. For specific code implementation, you can refer to the [wiki](https://wiki.seeedstudio.com/xiao_mg24_sense_built_in_sensor/#code-implementation).

is-qian, 2024-12-18:
In this [commit](https://github.com/Seeed-Studio/Seeed_Arduino_LSM6DS3/commit/68544d0088d36d6d1c74cc344a30503182a6a535), power control has been added, and the macro name has been changed. You can try again with the latest library. Thank you for your testing.

rei-vilo, 2024-12-18:
Solved with the commit, thank you!

closed 2024-12-18 by rei-vilo