Seeed-Studio/Seeed_Arduino_LSM6DS3 #23: Support for timestamps in FIFO

ScottMit, 2025-06-26:
Hi,

Thanks for supporting this library on Seeed devices.
I have a Xiao Sense board that is recording IMU data and I would like to record timestamps from the FIFO.
Could this functionality be added to the library?

Thanks.
Scott

baorepo, 2025-06-26:
@Yuan-AZKKXL 

**Background**
- The LSM6DS3 sensor does support FIFO timestamping, which records hardware timestamps alongside sensor data if enabled.
- This typically involves configuring the TIMESTAMP and FIFO registers so that timestamp values are included in the FIFO stream.

**To Implement This:**
1. Enable timestamp storage in FIFO by setting the appropriate registers (e.g., TIMESTAMP_CTRL and FIFO_CTRLx).
2. Update the library’s FIFO reading logic to recognize and parse timestamp tags and values from the FIFO stream.
3. Extend the library’s API so that timestamps can be accessed by users, along with accelerometer and gyroscope data.

**Reference:**  
- [LSM6DS3 Datasheet](https://www.st.com/resource/en/datasheet/lsm6ds3.pdf) (see FIFO and TIMESTAMP_CTRL sections).

**Example (Pseudo-Arduino code):**
```cpp
// Enable timestamp in FIFO
imu.writeRegister(LSM6DS3_REG_TIMESTAMP_CTRL, 0x80); // Enable timestamp
imu.writeRegister(LSM6DS3_REG_FIFO_CTRL4, 0x08); // Enable timestamp batching

// Read FIFO and decode timestamp pattern
while (fifoHasData()) {
    uint8_t tag = readFifoTag();
    if (tag == LSM6DS3_TAG_TIMESTAMP) {
        uint32_t ts = readTimestamp();
        Serial.println(ts);
    }
    // ... handle other tags (accel, gyro, etc)
}
```

**Next Steps:**
- Check if the current library already parses FIFO tags and timestamp registers.
- If not, we could propose or submit a PR to enable timestamping, parse the FIFO timestamps, and document the new API.

Yuan-AZKKXL, 2025-06-26:
@ScottMit 
hello,Do you want us to provide an interface implementation to activate the pedometer and timestamp data in the FIFO according to the steps in the picture?

[image]

ScottMit, 2025-06-26:
@Yuan-AZKKXL 
Hello. and thanks for the message.
We would like option 1, where data is written to the FIFO at the ODR_FIFO rate, but I imagine other people would be interested in option 2, where data is written for every new step count. So, an interface for either of these implementations would be good.
It looks like some of the steps already have an interface in the library: 1. turn on the accelerometer, 3. set fifo decimation, 7. configure the FIFO mode. maybe others.

Yuan-AZKKXL, 2025-06-26:
Is this the result you were looking for?
ΔT(ms) = (Current timestamp - Previous timestamp) × 0.625

[image]

Yuan-AZKKXL, 2025-06-27:
The approach you mentioned seems unfeasible because the timestamp data and the accelerometer/gyroscope data structures appear inconsistent. The current solution is to provide a separate read function for the FIFO timestamp, and you can enable or disable the timestamp functionality as done in the SparkFun_LSM6DS3_Arduino_Library.

[image]

ScottMit, 2025-06-27:
@Yuan-AZKKXL Thanks for the explanation and clear code example. 
I think reading the LSM6DS3_ACC_GYRO_TIMESTAMP0_REG gets the current timestamp, not the timestamp as recorded when data was passes to the FIFO.
You could test this by adding a small delay to the while loop and seeing if the timestamp delta changes. If it does, then it's not getting the timestamp from the FIFO.
For our use, we need timestamp data for the readings going into the FIFO because we need to check when readings are collected and if readings are being dropped.
To get FIFO timestamps the datasheet says you need to read this "as a 4th FIFO data set". I'm not exactly sure what that means, but was thinking it means the timestamp data is simply added to the FIFO data stream as a 4th option (after Acc, Gyro, & Mag) and you have to know the FIFO pattern in order to decipher it.
Let me know if my thinking is completely miss-guided.

Yuan-AZKKXL, 2025-06-30:
Yes, you're correct. The register LSM6DS3_ACC_GYRO_TIMESTAMP0_REG provides the current timestamp, not the one recorded when data was pushed into the FIFO. Regarding the timestamp being a "4th FIFO data set" as stated in the datasheet, it means the timestamp data is appended to the FIFO data stream alongside accelerometer, gyroscope, and external sensor data (note: the order here aligns with the datasheet's FIFO configuration). Thank you for the clarification—this makes the mechanism clear.

Here's the current implementation result; please verify if it meets your requirement for capturing FIFO entry timestamps to monitor data collection timing and potential drop-offs.

The difference between two FIFO timestamps: 9093440 - 9091891 = 1549.
Calculation:
1549×25μs=38725μs≈40ms

Initially, the timestamp frequency was assumed to be 25 Hz (since 
1/25Hz=40ms
), but according to the datasheet, the actual frequency should be 26 Hz.

[image]

ScottMit, 2025-06-30:
@Yuan-AZKKXL Thanks! That looks like what we are after. I will test it and let you know how we go.
Much appreciated.

ScottMit, 2025-07-08:
@AZKKXL. We have tested the timestamp code and its working well. Thanks heaps!

I do have a suggestion for how it is implemented. 
In the future someone is probably going to want to get step count from the FIFO buffer, in the current implementation the step count data is being discarded.

You could change the implementation to a two step process:
    - read the 4th data set with _fifoDataSetRead()_
    - extract time stamp data with _getTimestamp()_
  e.g.
`myIMU.getTimestamp(myIMU.fifoDataSetRead());`

For step count the implementation could be something like:
`myIMU.getStepCount(myIMU.fifoDataSetRead());`

Thanks again.
Scott

closed 2025-07-30 by github-actions[bot]

AZKKXL, 2025-08-07:
Hi @pikaro, regarding your confusion, I'd like to explain from the following points:

First, in the FIFO data of LSM6DS3, there are a total of four data sets, which you can verify in section 5.5.6 of the datasheet:

Group 1 FIFO data: Gyroscope data, 6 bytes
Group 2 FIFO data: Accelerometer data, 6 bytes
Group 3 FIFO data: External sensor data, 6 bytes
Group 4 FIFO data: External sensor data, or step counter and timestamp information, or temperature sensor data (6 bytes, including timestamp)

Second, although the 4th group data has 6 bytes, the timestamp data is 24-bit (datasheet section 8.8). Among them, byte 1 contains bits 8-15, byte 2 contains bits 16-23, byte 3 is empty, and byte 4 contains bits 0-7. The remaining bits are for step count data, but in the latest code, this data is discarded. That's why ScottMit mentioned, "In the future someone is probably going to want to get step count from the FIFO buffer; in the current implementation, the step count data is being discarded." (And I've recently been trying to fix this issue.) Therefore, a uint32_t is returned here, even though the timestamp data is only 24-bit (in fact, you can only return uint32_t since there's no uint24_t type).

Third, the function status_t LSM6DS3Core::readRegisterRegion(uint8_t* outputPointer, uint8_t offset, uint8_t length) takes a starting address and reads length bytes from that address onwards. So in my call, it starts from the address LSM6DS3_ACC_GYRO_FIFO_DATA_OUT_L and reads 3 * uint16_t bytes, which means reading an entire data set instead of repeatedly reading this register.

Fourth, since the FIFO is a first-in-first-out (FIFO) data set, you must read strictly according to these four groups. Even if you don't use the 3rd data set, you must read it first before you can obtain the timestamp data in the 4th data set. Failing to do so will result in complete garbling of your data.

Thanks for your comments. Feel free to provide further feedback if you have other questions.